raburton/rboot #8: Slow boot

slaff, 2016-03-18:
When using rBoot the booting time is roughly 7 times higher.

I used the Basic_Blink example in Sming. 
Without rBoot, the boot time is ~ 300 ms
With rBoot and with/or without RTC_ENABLED the boot time is ~2300  

Any ideas why is that happening ?

raburton, 2016-03-18:
The rBoot code has a 2 second delay to allow you to connect a terminal
before the messages are printed. This should be commented out by default,
but I could have accidentally committed it uncommented? rBoot doesn't
normally slow the boot by any significant amount and the fact it seems to
have done so here by exactly 2 seconds makes me suspect that deliberate
delay is in effect.

Richard.

slaff, 2016-03-18:
Got it! In rboot.c: 

https://github.com/SmingHub/Sming/blob/develop/Sming/rboot/rboot.c#L220

```
	// delay to slow boot (help see messages when debugging)
	ets_delay_us(2000000);
```

Will you change this or I should submit a PR?

raburton, 2016-03-18:
fixed in 4cf6132f6ee624318c04b2a8d9e37da64b7c283c

closed 2016-03-18 by raburton