raburton/rboot #17: How update rboot.bin over the air

fvpalha, 2016-07-01:
Hi.

I have some devices with rboot version v1.4.0.
I build a new firmware with rboot version v1.4.1 and I updated the **rom0** and **rom1** over the air.
But the devices with old rboot.bin version does not change SSID/Password for Station Mode with command "wifi_station_set_config".

Someone can help me?

fvpalha, 2016-07-04:
[image]

Comparing versions of rboot.bin 1.3.0 and 1.4.1 - no important difference.

fvpalha, 2016-07-04:
I do not understand what is wrong. Why the devices with new roms do not change the router target?

raburton, 2016-07-05:
I'm in Cambodia at the moment on holiday so can't help you much, but rBoot
does not affect what you can do with WiFi, or anything else, once the
application rom is booted.

fvpalha, 2016-07-05:
Thank you @raburton
A happy holiday to you.
I will identify the problem and I will find a solution.

fvpalha, 2016-07-12:
Hi @raburton 

I found the problem and the solution:
[Important SDK Patch Released](http://bbs.espressif.com/viewtopic.php?f=46&t=2407&sid=15ff5b9972035e8e05be64d3ef8cc8e9)

> SDK versions earlier than V1.5.4.1 have the limitation of flash map.  It is not clear that which sector of flash is available in users’ systems. Therefore the older SDK versions need to be upgraded.  In addition, “esp_init_data_default.bin” has been modified in this release.

raburton, 2016-07-23:
Sorry about the delay in getting back to you. Is everything working ok now? Is there anything I need to do to accommodate this SDK change?

fvpalha, 2016-07-25:
Hi @raburton ,

I found other case: [ESP SDK and RBoot Woes](http://tech.scargill.net/esp-sdk-and-rboot-woes/)

fvpalha, 2016-07-26:
Hi @raburton .

At moment I am using SDK 1.5.3 and rBoot [1.3.0](https://github.com/raburton/rboot/commit/af71a160558dd1f219c15ee74d9916b987102711#diff-00b4e5ab5436bd5bee4ce43ac1908a4a)  without problems.

Thank you.

raburton, 2016-07-27:
Tested sdk 1.5.4 this morning and rboot-sampleproject is building, running and OTAing just fine. Pete also has it working now too, he found some odd changes with the SDK he needed to account for, but nothing that needed changing in rBoot: http://tech.scargill.net/esp8266-home-control-update/

fvpalha, 2016-07-29:
Hi @raburton .

Now my code works with SDK 1.5.4.1 and rboot v1.4.1

But, the devices with previous roms does not works after remote update.
I am thinking of not maintaining legacy support.

closed 2016-09-03 by raburton