raburton/rboot #26: Add ESP32 Support

nitrotm, 2016-12-20:
As the ESP32 is available for some time now, that would be great to see how to port rboot to that chip if possible. My first impression is that it's pretty close to ESP8266 in terms of architecture and the RTOS sdk seems very similar in features / boot image format.
But there are two cores and new features so it might be just an impression.
Did someone try rboot on ESP32 already?

nitrotm, 2016-12-20:
Just saw espressif last sdk (esp-idf). It seems much more open than the ones for esp8266 (the bootloader source is available), so might not be so important to have rboot on the esp32 after all...

closed 2016-12-20 by nitrotm

raburton, 2016-12-21:
Agreed, I don't think there is the same need for rBoot on esp32 as there was on the original.