raburton/rboot #33: Potential for flash corruption...

snichols, 2017-02-24:
Here's a neat issue we've run into with our live devices using rboot.  If the device loses power when rboot is writing to flash, it's possible for write to be incomplete.  Basically if flash erase happens and power is lost before write then flash is left in the erased state.

It's a rare problem -- 1 : 1000 so far.  But it has happened.  Only way I can see this being worked around is to have two copies of the configuration and double-buffer them.

raburton, 2017-02-26:
That does sound rare, to hit a problem between the erase and write of the single config sector, but not possible. However, if the config sector is lost rBoot will create a new default one on next boot so it should then boot the first rom (this may not be the one you just flashed, but you'll be back up and running).

snichols, 2017-03-08:
Yeah, that's the fallback for sure... and works.  Just causes a double OTA which isn't the worst thing in 1:1000 chance. :)

closed 2017-03-08 by snichols