raburton/rboot #36: Issue with mapping of 1MB ROM

ukrsms, 2018-03-26:
Earlier I have been used rBoot and two 512k+512k ROM OTA. However now 512kB ROM is not enough for our project, so I tried to use RBOOT_BIG_FLASH case. I changed the makefile in order to weak Cache_Read_Enable_New() and add the alternative one.
However, if I flash ROM0 and ROM1 (at 0x2000 and 0x102000 respectively) and command rBoot to start ROM1 it does start and work fine. However, if I remove ROM0 the firmware crashes after some time after starting. It seems that there is some problem with mapping because of the issue occurs in the case when ROM0 and ROM1 differs.
Could anybody explain this issue?

I'm using RTOS SDK1.5 and I didn't try this with RTOS SDK1.3-1.4.

ukrsms, 2018-03-26:
I have already found the problem. It was caused by that I store text data in flash and read it out with
> static const char flash_str[] ICACHE_RODATA_ATTR STORE_ATTR = FLASH_STR;
> spi_flash_read((u32)flash_str-0x40200000, str_tmp, strlen(flash_str)+1);
So I just need to "remap" the address to read from.

closed 2018-03-26 by ukrsms

raburton, 2018-03-26:
Glad you got it sorted!

fvpalha, 2018-08-10:
@ukrsms can you detail the solution?

I have the same problem with the NONOS-SDK 2.2.1.
When the current rom is ROM1 (0x102000) the update process fail. The ROM0 (0x2000) is erased until the address 0x3000 and the process is interrupted.

@raburton do you have a tip?