raburton/rboot #50: RBOOT SDK 3.1

scargill, 2019-03-31:
Having finally gotten RBOOT and my ESP-GO code running on ESP SDK 3.1 - I'm occasionally getting E.M.5832 - which results in a safe OTA failure... next time it usually works.... is there anything I can do to narrow down why I'm sometimes seeing this memory error?

uzi18, 2019-03-31:
I think it is related to low amount free memory, so can't allocate 5832 bytes and error shows.

scargill, 2019-03-31:
This is weird... it works many 50% of the time and Richard suggested fragmentation, then, just half an hour ago I discovered a 4K buffer in my main code page (not in a function) which is not in use, so deleted it and recompiled - taking my heap reading from 7k+ to 11k+ - surely this would minimise any issues - yet the error is occurring just the same - and sill only some of the time I OTA. Getting rid of that unused array has made no difference.... E:M 5732 and E:M 5864 - other times I try OTA (same code, same board) I get none of that and it works,

raburton, 2019-04-01:
It wouldn't take a lot of heap fragmentation to prevent you getting an allocation of 4k from 11k. Two single badly placed bytes in the heap would be sufficient to prevent it. You need to find out where the offending malloc is and see if you can use less memory there. It may not be in rBoot anyway. If it is, my guess would that it's in rboot_write_flash - in that case try sending it smaller chunks of data to write.

scargill, 2019-04-01:
I’ve never really gone into RBOOT in depth, in the rboot header there is a reference to sector size at 4K – not sure how to safely manipulate that.

 

Is It possible, before the defrag starts, to do some kind of general RAM defrag?

 

I use very little MALLOC, preferring (for no special reason) local variables as needed. In fact, in my main page, malloc does not occur at all.

 

Pete

raburton, 2019-04-03:
Closed as duplicate

closed 2019-04-03 by raburton