raburton/rboot #51: RBOOT NONOS SDK 3.1

scargill, 2019-04-17:
As original issue seems closed, you may recall you kindly re-wrote the flash-write section of RBOOT to get over the unreliability I was (am) suffering only with RBOOT or rather with OTA using SDK 3.1 - the changes produced a reboot on use. Did you get a chance to look at this?

I put in your new flash write routine (a couple of weeks ago, I've been travelling since so sorry if I missed anything) and on trying OTA, the Raspberry Pi 3 crashed and rebooted immediately. It does this every time.

For clarity, the originsl code works perfectly every time with SDK 2.1

With SDK 3.1 all aspects of my code work perfwctly except RBOOT/SDK which works sometimes and fails other times. With the new flash write code, OTA attempts produce an immediate reboot every time.

Pete

raburton, 2019-04-17:
The raspberry pi crashed when you do ota? Not the esp? Where does a pi come into it?
I looked at it, I can't find anything wrong with the new code and can't replicate the crash. Perhaps if it's consistent on yours you could add some debug output to the serial console and identify which line of code is causing the crash and the values of the values it's working with at that point and I will try to see if I can figure out what's happening at your end.

scargill, 2019-04-17:
Sorry.. esp crashes immediately when I try ota with the new function.

scargill, 2019-04-17:
Will do my best. Weather will be rubbish for the rest of the week so I will 
give it a go.

scargill, 2019-05-10:
Still no-where with SDK 3.1 - the original RBOOT (on for example ESP12) works some of the fime, failing to OTA other times, the changed version (flash writing routine update) crashes completely when OTA is attempted. Meanwhile on SDK 2.1, all is completely reliable.

raburton, 2019-05-10:
I know, did you get the requested debug?

scargill, 2019-05-12:
I will try to get that.