raburton/rboot #53: Error in checksum calculation?

gpreviato, 2019-08-02:
Hi,
I had worked a lot you your great code (but also I've made some mess in...), and I was looking to what seem be a miscalculation in the checksum. Actually I have a mismatch between the checksum calculated by rboot and the one calculated from esptool.py. Seems the latter doesn't consider the first segment in checksum calculation. Here the results from a modified esptool (for log):
./esptool.py --chip esp8266 image_info ../../firmware/firmware.bin
Entry point: 4010023c
4 segments

Segment 1: len 0xc1530 load 0x00000000 file_offs 0x00000008
Segment 2: len 0x077b4 load 0x40100000 file_offs 0x000c1548
Segment 3: len 0x00cf4 load 0x3ffe8000 file_offs 0x000c8d04
Segment 4: len 0x019bc load 0x3ffe8cf8 file_offs 0x000c9a00
Seg: 0 -> seg.include_in_checksum: 0. DataLen: 0xc1530
Checksum: 0xef
Seg: 1 -> seg.include_in_checksum: 1. DataLen: 0x77b4
Checksum: 0x56
Seg: 2 -> seg.include_in_checksum: 1. DataLen: 0xcf4
Checksum: 0x66
Seg: 3 -> seg.include_in_checksum: 1. DataLen: 0x19bc
Checksum: 0x76
Checksum: 76 (valid)

As you can see the first segment is excluded from calculation.
Doing the same with your code, i.e. adding:
			if (sectcurrent!=0)
			{
				for (loop = 0; loop < readlen; loop++) {
					chksum ^= buffer[loop];
				}
			}
seems set everything ok.

What do you think?

Rgds,

--gra

gpreviato, 2019-08-04:
> Hi,
> I had worked a lot you your great code (but also I've made some mess in...), and I was looking to what seem be a miscalculation in the checksum. Actually I have a mismatch between the checksum [...]

Do not mind about this... I've to go digger into....

raburton, 2019-08-04:
Sorry I haven't had chance to look at this, I'm not sure which checksum you're referring too. If it's the standard one then it must be right or the esp wouldn't run the code. if it's the extra rBoot checksum, as far as I know this is not calculated by esptool.py.