raburton/rboot #57: 4MB vs 1MB - simple blink app rom switch

kiralikbeyin, 2019-08-22:
I have 2 simple blink app which switches roms.

Nodemcu works fine but same blink apps dont work on esp01 or sonoff dualr2

flash for 32m (4MB) with qio - blink 2 is on 0x202000 
````
esptool.py -p /dev/cu.SLAB_USBtoUART --baud 115200 write_flash -fs 4MB -fm qio -ff 40m \
		0x0 ../../bootloader/firmware/rboot.bin 0x2000 ./firmware/blink.bin  0x202000 ../../bootloader/firmware_prebuilt/blink.bin
````

flash for 8m (1MB) with dout - blink 2 is on 0x82000 
````
esptool.py -p /dev/cu.usbserial-A7012RPU --baud 115200 write_flash -fs 1MB -fm dout -ff 40m \
		0x0 ../../bootloader/firmware/rboot.bin 0x2000 ./firmware/blink.bin  0x82000 ../../bootloader/firmware_prebuilt/blink.bin
`````

````
ESP01 SERIAL

ESP-Open-SDK ver: 0.0.1 compiled @ Aug 22 2019 13:06:16
phy ver: 273, pp ver: 8.3

mode : softAP(62:01:94:9e:85:88)
add if1
bcn 100
BLINK BLINK say : 0
BLINK BLINK say : 1
BLINK BLINK say : 2
BLINK BLINK say : 3
BLINK BLINK say : 4
BLINK BLINK say : 5
BLINK BLINK say : 6
BLINK BLINK say : 7
blınk magic : 225
version : 1
mode : 0
current_rom : 0
gpio_rom : 0
count : 2
blınk rboot_get_current_rom : 0
blınk SAVED   rboot_set_config
blınk rboot_get_current_rom : 1
bcn 0
del if1
usl
sul 0 0

ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x3ffe8000, len 680, room 16
tail 8
chksum 0x40
load 0x40100000, len 1548, room 0

0x40100000: _stext at ??:?

tail 12
chksum 0x37
csum 0x37

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DOUT
Flash Speed:  40 MHz
rBoot Option: RTC data

Booting rom 1 at 82000, load addr b86a0.
//STOPS HERE BUT NODEMCU CONTINUE TO ROM 0 
````

blink 1
````
#include <stdlib.h>
#include "espressif/esp_common.h"
#include "esp/uart.h"
#include "FreeRTOS.h"
#include "task.h"
#include "esp8266.h"

#include "rboot-integration.h"
#include <rboot-api.h>

const int gpio = 13;

void blinkenTask(void *pvParameters)
{
  int say = 0;

    gpio_enable(gpio, GPIO_OUTPUT);
    while(1) {

      printf("BLINK BLINK say : %d \n", say);

      gpio_write(gpio, 1);
      vTaskDelay(600 / portTICK_PERIOD_MS);
      gpio_write(gpio, 0);
      vTaskDelay(600 / portTICK_PERIOD_MS);

      say ++;
      if(say>7 )
      {
        rboot_config conf;
        conf = rboot_get_config();

        printf("blınk magic : %d \n", conf.magic);
        printf("version : %d \n", conf.version);
        printf("mode : %d \n", conf.mode);
        printf("current_rom : %d \n", conf.current_rom);
        printf("gpio_rom : %d \n", conf.gpio_rom);
        printf("count : %d \n", conf.count);
        printf("blınk rboot_get_current_rom : %d \n", rboot_get_current_rom());

        if(rboot_get_current_rom())
        {
          conf.current_rom = 0;
        }
        else
        {
          conf.current_rom = 1;
        }
        if(rboot_set_config(&conf))
        {
          printf("blınk SAVED   rboot_set_config   \n");
        }
        else {
          printf("blınk FAILED   rboot_set_config   \n");
        }

            printf("blınk rboot_get_current_rom : %d \n", rboot_get_current_rom());
            sdk_system_restart();
      }

  } //while END
}

void blinkenRegisterTask(void *pvParameters)
{
    GPIO.ENABLE_OUT_SET = BIT(gpio);
    IOMUX_GPIO2 = IOMUX_GPIO2_FUNC_GPIO | IOMUX_PIN_OUTPUT_ENABLE; /* change this line if you change 'gpio' */
    while(1) {
        GPIO.OUT_SET = BIT(gpio);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        GPIO.OUT_CLEAR = BIT(gpio);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

void user_init(void)
{
    uart_set_baud(0, 115200);
    xTaskCreate(blinkenTask, "blinkenTask", 256, NULL, 2, NULL);
    //xTaskCreate(blinkenRegisterTask, "blinkenRegisterTask", 256, NULL, 2, NULL);
}
````

blink 2
````
#include <stdlib.h>
#include "espressif/esp_common.h"
#include "esp/uart.h"
#include "FreeRTOS.h"
#include "task.h"
#include "esp8266.h"

#include "rboot-integration.h"
#include <rboot-api.h>
 
const int gpio = 13;

void blinkenTask(void *pvParameters)
{
  int say = 0;

    gpio_enable(gpio, GPIO_OUTPUT);
    while(1) {
      printf("OK OK OK ");
      printf("BLINK2 say : %d \n", say);

      gpio_write(gpio, 1);
      vTaskDelay(2000 / portTICK_PERIOD_MS);
      gpio_write(gpio, 0);
      vTaskDelay(2000 / portTICK_PERIOD_MS);

      say ++;

      if(say>9 )
      {
              rboot_config conf;
              conf = rboot_get_config();

              printf("BLINK2 magic : %d \n", conf.magic);
              printf("version : %d \n", conf.version);
              printf("mode : %d \n", conf.mode);
              printf("current_rom : %d \n", conf.current_rom);
              printf("gpio_rom : %d \n", conf.gpio_rom);
              printf("count : %d \n", conf.count);
              printf("BLINK2 rboot_get_current_rom : %d \n", rboot_get_current_rom());

              if(rboot_get_current_rom())
              {
                conf.current_rom = 0;

              }
              else
              {
                conf.current_rom = 1;

              }

              if(rboot_set_config(&conf))
              {
                printf("BLINK2 SAVED   rboot_set_config   \n");
              }
              else {
                printf("BLINK2 FAILED   rboot_set_config   \n");
              }

                  printf("BLINK2 rboot_get_current_rom : %d \n", rboot_get_current_rom());
                  sdk_system_restart();

        }

    }
}

void blinkenRegisterTask(void *pvParameters)
{
    GPIO.ENABLE_OUT_SET = BIT(gpio);
    IOMUX_GPIO2 = IOMUX_GPIO2_FUNC_GPIO | IOMUX_PIN_OUTPUT_ENABLE; /* change this line if you change 'gpio' */
    while(1) {
        GPIO.OUT_SET = BIT(gpio);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        GPIO.OUT_CLEAR = BIT(gpio);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

void user_init(void)
{
    uart_set_baud(0, 115200);
    xTaskCreate(blinkenTask, "blinkenTask", 256, NULL, 2, NULL);
    //xTaskCreate(blinkenRegisterTask, "blinkenRegisterTask", 256, NULL, 2, NULL);

}
````

Is there a special file (eagle.rom.addr.v6.ld -rboot-stage2a.ld -eagle.app.v6.ld) for 1MB ?

kiralikbeyin, 2019-08-22:
rboot-stage2a.ld https://github.com/raburton/rboot/blob/master/rboot-stage2a.ld
````
MEMORY
{
  dport0_0_seg :                        org = 0x3FF00000, len = 0x10
  dram0_0_seg :                         org = 0x3FFE8000, len = 0x14000
  **iram1_0_seg :                         org = 0x4010FC00, len = 0x400**
  irom0_0_seg :                         org = 0x40282010, len = 0x3C000
}
`````
iram1_0_seg is different in rboot lib,
sample have   ```iram1_0_seg :                         org = 0x40100000, len = 0x8000``` in both rom0.ld and rom1.ld

Also i didnt understand len = 0x3C000 my real binary will be 670k for rom0 and 260k for rom1 do i have to change 0x3C000 ? 
(My test blink files are only 255k)

Note that eagle.rom.addr.v6.ld file has extra rows in sample:
PROVIDE ( Enable_QMode = 0x400044c0 );

PROVIDE ( Td0 = 0x3fffd100 );
PROVIDE ( Td4s = 0x3fffd500);
PROVIDE ( rcons = 0x3fffd0f0);
Do i have to add these also?

kiralikbeyin, 2019-08-22:
eagle.app.v6.ld file is now
```
MEMORY
{
  dport0_0_seg :                        org = 0x3FF00000, len = 0x10
  dram0_0_seg :                         org = 0x3FFE8000, len = 0x14000
  iram1_0_seg :                         org = 0x40100000, len = 0x8000
  irom0_0_seg :                         org = 0x40202010, len = 0x3C000
}
```

rboot-stage2a.ld changed **iram1_0_seg**:
````
MEMORY
{
  dport0_0_seg :                        org = 0x3FF00000, len = 0x10
  dram0_0_seg :                         org = 0x3FFE8000, len = 0x14000
  iram1_0_seg :                         org = 0x4010FC00, len = 0x400
  irom0_0_seg :                         org = 0x40282010, len = 0x3C000
}
````
Stops at as before
Booting rom 1 at 82000, load addr b86a0.
//STOPS HERE BUT NODEMCU CONTINUE TO ROM 0 

rboot-stage2a.ld changed **iram1_0_seg** didnt make any boot  :
````
MEMORY
{
  dport0_0_seg :                        org = 0x3FF00000, len = 0x10
  dram0_0_seg :                         org = 0x3FFE8000, len = 0x14000
  iram1_0_seg :                         org = 0x40100000, len = 0x8000
  irom0_0_seg :                         org = 0x40282010, len = 0x3C000
}
````

kiralikbeyin, 2019-08-22:
@raburton i am changing eagle.rom.addr.v6.ld -rboot-stage2a.ld -eagle.app.v6.ld files
but there is a folder : https://github.com/SuperHouse/esp-open-rtos/tree/master/ld
Does it need action?

I copied rboot-sample files to rboot
https://github.com/raburton/rboot-sample/blob/master/eagle.rom.addr.v6.ld to https://github.com/raburton/rboot/blob/master/eagle.rom.addr.v6.ld 

https://raw.githubusercontent.com/raburton/rboot-sample/master/rom0.ld to https://github.com/raburton/rboot/blob/master/eagle.app.v6.ld

https://raw.githubusercontent.com/raburton/rboot-sample/master/rom1.ld to
https://github.com/raburton/rboot/blob/master/eagle.app.v6.ld

and maked irom0_0_seg len = 0x3C000 to len = 0x50000

didnt make any boot ..

raburton, 2019-08-22:
> rboot-stage2a.ld https://github.com/raburton/rboot/blob/master/rboot-stage2a.ld

You don't need to change any of the rBoot ld files (such as rboot-stage2a.ld), in fact you should not be touching these at all.

You need to change the ld files you use to link your own application. You normally only have one of these, but you need a second one with a different irom0_0_seg org address. See the example in rboot-sample. Then you need to link your application twice, once with each of these lib files, and flash the correct one to the correct half of the flash.

> I copied rboot-sample files to rboot

Why did you do that? rboot repo contains the rBoot boot loader, rboot-sample is a demo user application that can be run with rBoot that demonstrates OTA. You need to build both, separately, not randomly copy files from one to the other. I suggest you read the instructions in each repository and build them to get rBoot and the sample application running before you try using it with your own application. That way you will hopefully gain some understanding of what is actually happening in the boot process. You might also want to read the explanatory entries in my blog.

kiralikbeyin, 2019-08-22:
I understand it now! i made a duplicate of this ld file https://github.com/SuperHouse/esp-open-rtos/blob/master/ld/program.ld and i only changed 

irom0_0_seg : org = 0x40202010, len = 0x50000
irom0_0_seg : org = 0x40282010, len = 0x50000

add parameters.mk
```
ifdef r1
LINKER_SCRIPTS += $(ROOT)ld/rom.ld $(ROOT)ld/pr1.ld
endif

ifdef r2
LINKER_SCRIPTS += $(ROOT)ld/rom.ld $(ROOT)ld/pr2.ld
endif

#LINKER_SCRIPTS += $(ROOT)ld/program.ld $(ROOT)ld/rom.ld
````
prepare bins
make r1=1 for blink1 app
make r2=1 for blink2 app

flash it
esptool.py -p /dev/cu.wchusbserial1410 --baud 115200 write_flash -fs 1MB -fm dout -ff 40m \
		0x0 ../../bootloader/firmware/rboot.bin 0x2000 ./firmware/blink.bin  0x82000 ../../bootloader/firmware_prebuilt/blink.bin

Now i will try this:
rom0 670k bin
rom1 250k OTA updater bin

closed 2019-08-22 by kiralikbeyin