raburton/rboot #58: Can this codebase be used with ESP8266_NON_OS SDK for OTA ?  I am using 16 MBits flash and have two OTA partitions ,

Dipendra77, 2020-03-04:


raburton, 2020-03-04:
Yes, it can be used for nonos, in fact when it was first written that's all there was. rBoot itself does not need the sdk to compile or the use any sdk code to run. Code that runs in you rom (e.g. downloads and writes to flash) will need to use SDK code. An example application that includes sample OTA code can be found here: https://github.com/raburton/rboot-sample-bigflash
If you are not actually using rBoot bootloader, but trying to use rBoot OTA code (sounds a bit like that from your question) then that won't work. As you say there will be no rBoot config to read and without bootloader there will be no way to start the second rom after an OTA update.

Dipendra77, 2020-03-08:
Is there any way to make OTA work with rboot and ESP8266 NON OS SDK ?

raburton, 2020-03-08:
I haven't tested with that specific version of the SDK, but rBoot should be able to boot any rom regardless of what SDK (or otherwise) it was built with.

fvpalha, 2020-03-09:
I use rboot with ESP8266 NONOS SDK v3.1.0-dev
commit: [3783b9d1](https://github.com/espressif/ESP8266_NONOS_SDK/commit/3783b9d10806e05016f1102648e7763e68f1c6c9) (Dec 4, 2019)