raburton/rboot #59: GPIO Question

HomeHomeApp, 2020-05-17:
@raburton 
Is it possible to make a gpio high or low when esp8266 starts?

raburton, 2020-05-17:
Yes it's possible, but not built in - you'd need to add a little code to rBoot to achieve it. Of course rBoot quickly hands control over to the user rom (and all it's SDK code), so I don't know how long any change you make to a GPIO will remain, the SDK could reset it again. I don't know the exact code you'd need to set the GPIO in baremetal mode now, it's been a long time since I looked at this, but have a look at the code that handles checking a GPIO on startup and you'll probably work out what you need to do. See the couple of functions from here https://github.com/raburton/rboot/blob/master/rboot.c#L130

HomeHomeApp, 2020-05-17:
I understood rboot working process in 2 months (never tried a bootloader before)

Your lib is awesome, i am using it with esp-open-rtos. 2 roms configured in 1mb flash (750k+250k) and both roms can be updated with ota. 

I think WRITE_PERI_REG will be used but these codes are very complex for me

My problem is relay 1,2,3 are high untill sketch begins.
      relay_gpio_1 = 12;
      relay_gpio_2 = 5;
      relay_gpio_3 = 4;
      relay_gpio_4 = 15;

Is there any easy way to make 12,5,4 to low?

raburton, 2020-05-17:
As I say, I don't know the exact code you'll need (because I haven't looked at GPIO stuff for several years), but you can try writing something based on the code in the function https://github.com/raburton/rboot/blob/master/rboot.c#L162. This checks the state of a GPIO, by pulling it up and then checking if it's been pulled back down by an external connection to ground. So it's setting the state of the pin to high and using it an an input, you want to do the opposite and set it to low as an output. Most of the code to do that will be the same, you'll just need to tweak it a little bit (sorry, but how to do that is the bit you'll need to work out).

HomeHomeApp, 2020-05-17:
Ok, i will work on this today, if i can find a solution i will write here. 
Thank you. 👍