raburton/rboot #62: ESP8266 disabling UART, Boot from SD, Jtag pins through rboot agains copy protection

aldock, 2023-01-17:
Hi,

I´m trying to disable ESP8266 pins, as I´m only need SCL/SDA, in order to avoid flash code copy.  Is rboot able to do this ?  or it is not possible.  *Note:  I´ll use ESP8285, internal flash, to avoid external access.   Thank you!

raburton, 2023-01-17:
No, there is no way to protect the code on an esp8266 as it's stored on an external flash. I'm not familiar with the esp8285, but if it has built in flash then it will almost certainly have a config option to prevent reading out, with no need to try and disable pins.

aldock, 2023-01-17:
Thank you. Right, is hasn´t.  May be using 6v at the pins :) to blow the pins.  I think the way would be by software.

raburton, 2023-01-17:
The esp has to read from the flash, so if you kill the pins on the flash you'll probably not be able to boot the device, unless it has separate connections (even then you might damage them).

aldock, 2023-01-17:
Not the Esp8285 flash that is internal but the external pins, Ex. Gpio1(Tx) and Gpio12(Jtag TDI)

raburton, 2023-01-17:
As long as they don't just connect to the same place inside the package - the flash probably only has one set of data pins, so same ones may be connected to the esp as well as the external pins. Anyway, you can always test that out.
But there is nothing you can do with rBoot to disable this, and if espressif made any attempt to allow the flash to be protected, then I doubt there is anything you can do with software at all.

aldock, 2023-01-17:
Yes, that is right.  Ok, thank you.

aldock, 2023-01-17:
Just remembered..   Keeloc can do this.   https://keeloc.com
"Finally we wish to talk how to seal and protect via pin destruction of the USB serial pin port. This is done by writing firmware that every time it first boots ups the ESP8266 will turn high in voltage value and low impedance (the serial pins) at the same time shorting out these pins by block circuit to ground"

raburton, 2023-01-17:
Sounds a lot like snake oil to me. Their protection is light on detail but seems to be based on messing with the serial port pins when the bootloader is run - so maybe that does do what you wanted, but doesn't help - if you want to steal the code, just interrupt the boot before the bootloader runs. As for their encrypted rom, this is decrypted by the bootloader, which itself will be small and simple, so should be pretty easy to decompile and crack. Nothing your average Chinese cloner couldn't manage in a matter of minutes I'm sure.

aldock, 2023-01-17:
Really surprising, any way Espressif should make things easier.  Thank you very much.