raburton/rboot #63: ElF file bootrom.bin "no such file or directory: bootrom.bin" Ubuntu

aldock, 2023-01-23:
Hi Richard,

Hope you can help.
I´m trying to generate ELF file from my code, to find out any breach on it, following the orientation below
OK: import esp_bin2elf
Ok: import flash_layout
Ok: flash_layout = flash_layout.layout_without_ota_updates
Ok: rom = esp_bin2elf.parse_rom('flashdump.bin', 'path/to/flashdump.bin', flash_layout)
Ok: section_names = esp_bin2elf.name_sections(rom)
Fail: elf = esp_bin2elf.convert_rom_to_elf(rom, section_names, 'flash_bin.elf')
error: "no such file or directory: bootrom.bin"

I have changed directory, permissions etc, but no one resolved.]
May i use bootrom.bin from the code or should i generate a new one ?
Than you!

aldock, 2023-01-23:
Got it resolved, now it shows?
elf = esp_bin2elf.convert_rom_to_elf(rom, section_names, 'flash_bin.elf')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "esp_bin2elf.py", line 47, in convert_rom_to_elf
    elf_section = ElfSection(name, section.address, section.contents)
  File "esp_elf.py", line 121, in __init__
    raise Exception("can't find settings for %x" % (section_adddress))
NameError: global name 'section_adddress' is not defined

aldock, 2023-01-23:
First, just worked with python2 and section addres error.
with zayfod changes, works with python3 and i did the alternation from python2 to 3, all modules reinstalled again, elffile2..., found right path with os.getcwd(), running again.
Again all ok, except, last commmand, now line 122:

elf = esp_bin2elf.convert_rom_to_elf(rom, section_names, 'flash_bin.elf')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/osboxes/Desktop/esp_bin2elf.py", line 48, in convert_rom_to_elf
    elf_section = ElfSection(name, section.address, section.contents)
  File "/home/osboxes/Desktop/esp_elf.py", line 122, in __init__
    raise Exception("Can't find settings for 0x%08x" % section_address)
Exception: Can't find settings for 0x40200000

raburton, 2023-01-23:
I'm sorry but I have no idea what you're trying to do or how it's related to rBoot. It looks like you're using some tooling I've never seen before, perhaps the author of these python tools would be better able to answer your question.

aldock, 2023-01-23:
Thank you !  I´ll ask him.

aldock, 2023-01-25:
Found a way!  got the ELF directly from Arduino build :).  Super easy.  Thank you.

aldock, 2023-01-31:
Hi Richard.  I´m still pushing to minimize hacker chances of copy code.
I think one effort is use flash and chip unique ids along the code, may be in four places into the code, if not equal apply ESP restart ou while(1). Also a second step would be use your rboot solution but "encrypting". I tryied for 3 days folow the steps  here, https://richard.burtons.org/2015/05/22/a-boot-loader-tutorial-for-esp8266-using-rboot/,   but stopped when generating two elfs, i´ll still try.   Well, now i´m trying to find a way for encryptyed bootloader, the same as keeloc did and they probably used your development.   Thank you

raburton, 2023-01-31:
You could easily encrypt the iram section, and have rboot decrypt them on startup, perhaps based on the serial number (if you can easily access it in baremetal mode). But the rboot code that does the decryption can't be encrypted, so a hacker could easily decompile rboot and see exactly how the decryption is done, and then remove the encryption. It would prevent home cloning by the average user, but offers no serious protection.

aldock, 2023-02-01:
Ok thank you. I´ll try.

aldock, 2023-02-03:
Richard, with this solution is possible to encrypt the flash content ?   thank you

raburton, 2023-02-03:
Not code on flash. irom segments are not loaded to ram, such that you could
decrypt them when loaded. The flash is memory mapped, so all code being
executed from flash by this method needs to be stored on the flash
unencrypted.